ChenHoy/DROID-Splat #4: DROID-Splat Added to awesome-Implicit-NeRF-SLAM Repo

DoongLi, 2024-11-27:
Hi, this message serves as a reminder that your paper ”DROID-Splat: Combining end-to-end SLAM with 3D Gaussian Splatting“, has been included in the repository. https://github.com/DoongLi/awesome-Implicit-NeRF-SLAM.
If there is any missing information about your work on this repository (including paper links, code, datasets, website, etc), please let us know or submit a pull request (PR) directly.

ChenHoy, 2024-11-27:
Hi, so the missing paper link is now added. Can you include this or do I have to open a pull request?

DoongLi, 2024-11-27:
I have added it. DROID-Splat: Combining end-to-end SLAM with 3D Gaussian Splatting, arXiv, 2024. [[Paper](https://arxiv.org/pdf/2411.17660)] [[Code](https://github.com/ChenHoy/DROID-Splat)]

closed 2024-11-27 by ChenHoy