ChenHoy/DROID-Splat #7: Command '['ninja', '-v']' returned non-zero exit status 1.

JV-X, 2024-12-02:
Hello, I tried to build the DROID-Splat environment in my WSL2 subsystem. When I executed the ```python setup.py install``` command, I got the ```subprocess.CalledProcessError: Command '['ninja', '-v']' returned non-zero exit status 1.``` error

The log is too long, so some parts are omitted,full log in here :
[log.log](https://github.com/user-attachments/files/17972106/log.log)

key logs: 
```
(droidsplat) hygx@hygx:~/code/DROID-Splat$ python setup.py install
running install
/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/setuptools/_distutils/cmd.py:66: EasyInstallDeprecationWarning: easy_install command is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` and ``easy_install``.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://github.com/pypa/setuptools/issues/917 for details.
        ********************************************************************************

!!
  self.initialize_options()
running bdist_egg
running egg_info
writing droid_backends.egg-info/PKG-INFO
writing dependency_links to droid_backends.egg-info/dependency_links.txt
writing top-level names to droid_backends.egg-info/top_level.txt
reading manifest file 'droid_backends.egg-info/SOURCES.txt'
adding license file 'LICENSE'
writing manifest file 'droid_backends.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-x86_64/egg
running install_lib
running build_ext
/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/utils/cpp_extension.py:424: UserWarning: There are no g++ version bounds defined for CUDA version 12.1
  warnings.warn(f'There are no {compiler_name} version bounds defined for CUDA version {cuda_str_version}')
building 'droid_backends' extension
creating /home/hygx/code/DROID-Splat/build/temp.linux-x86_64-cpython-311/src/lib
Emitting ninja build file /home/hygx/code/DROID-Splat/build/temp.linux-x86_64-cpython-311/build.ninja...
Compiling objects...
Allowing ninja to set a default number of workers... (overridable by setting the environment variable MAX_JOBS=N)
[1/4] /usr/local/cuda-12.1/bin/nvcc  -I/home/hygx/code/DROID-Splat/thirdparty/eigen -I/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/include -I/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/include/torch/csrc/api/include -I/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/include/TH -I/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/include/THC -I/usr/local/cuda-12.1/include -I/home/hygx/anaconda3/envs/droidsplat/include/python3.11 -c -c /home/hygx/code/DROID-Splat/src/lib/correlation_kernels.cu -o /home/hygx/code/DROID-Splat/build/temp.linux-x86_64-cpython-311/src/lib/correlation_kernels.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -O3 -gencode=arch=compute_60,code=sm_60 -gencode=arch=compute_61,code=sm_61 -gencode=arch=compute_70,code=sm_70 -gencode=arch=compute_75,code=sm_75 -gencode=arch=compute_80,code=sm_80 -gencode=arch=compute_86,code=sm_86 -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=droid_backends -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++17
FAILED: /home/hygx/code/DROID-Splat/build/temp.linux-x86_64-cpython-311/src/lib/correlation_kernels.o
/usr/local/cuda-12.1/bin/nvcc  -I/home/hygx/code/DROID-Splat/thirdparty/eigen -I/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/include -I/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/include/torch/csrc/api/include -I/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/include/TH -I/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/include/THC -I/usr/local/cuda-12.1/include -I/home/hygx/anaconda3/envs/droidsplat/include/python3.11 -c -c /home/hygx/code/DROID-Splat/src/lib/correlation_kernels.cu -o /home/hygx/code/DROID-Splat/build/temp.linux-x86_64-cpython-311/src/lib/correlation_kernels.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -O3 -gencode=arch=compute_60,code=sm_60 -gencode=arch=compute_61,code=sm_61 -gencode=arch=compute_70,code=sm_70 -gencode=arch=compute_75,code=sm_75 -gencode=arch=compute_80,code=sm_80 -gencode=arch=compute_86,code=sm_86 -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=droid_backends -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++17
Killed
[2/4] c++ -MMD -MF /home/hygx/code/DROID-Splat/build/temp.linux-x86_64-cpython-311/src/lib/droid.o.d -pthread -B /home/hygx/anaconda3/envs/droidsplat/compiler_compat -DNDEBUG -fwrapv -O2 -Wall -fPIC -O2 -isystem /home/hygx/anaconda3/envs/droidsplat/include -fPIC -O2 -isystem /home/hygx/anaconda3/envs/droidsplat/include -fPIC -I/home/hygx/code/DROID-Splat/thirdparty/eigen -I/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/include -I/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/include/torch/csrc/api/include -I/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/include/TH -I/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/include/THC -I/usr/local/cuda-12.1/include -I/home/hygx/anaconda3/envs/droidsplat/include/python3.11 -c -c /home/hygx/code/DROID-Splat/src/lib/droid.cpp -o /home/hygx/code/DROID-Splat/build/temp.linux-x86_64-cpython-311/src/lib/droid.o -O3 -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=droid_backends -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++17
[3/4] /usr/local/cuda-12.1/bin/nvcc  -I/home/hygx/code/DROID-Splat/thirdparty/eigen -I/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/include -I/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/include/torch/csrc/api/include -I/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/include/TH -I/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/include/THC -I/usr/local/cuda-12.1/include -I/home/hygx/anaconda3/envs/droidsplat/include/python3.11 -c -c /home/hygx/code/DROID-Splat/src/lib/altcorr_kernel.cu -o /home/hygx/code/DROID-Splat/build/temp.linux-x86_64-cpython-311/src/lib/altcorr_kernel.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -O3 -gencode=arch=compute_60,code=sm_60 -gencode=arch=compute_61,code=sm_61 -gencode=arch=compute_70,code=sm_70 -gencode=arch=compute_75,code=sm_75 -gencode=arch=compute_80,code=sm_80 -gencode=arch=compute_86,code=sm_86 -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=droid_backends -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++17
/home/hygx/code/DROID-Splat/src/lib/altcorr_kernel.cu: In lambda function:
/home/hygx/code/DROID-Splat/src/lib/altcorr_kernel.cu:309:43: warning: ‘at::DeprecatedTypeProperties& at::Tensor::type() const’ is deprecated: Tensor.type() is deprecated. Instead use Tensor.options(), which in many cases (e.g. in a constructor) is a drop-in replacement. If you were using data from type(), that is now available from Tensor itself, so instead of tensor.type().scalar_type(), use tensor.scalar_type() instead and inst [... cut ...]
  File "/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/utils/cpp_extension.py", line 1774, in _write_ninja_file_and_compile_objects
    _run_ninja_build(
  File "/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/utils/cpp_extension.py", line 2116, in _run_ninja_build
    raise RuntimeError(message) from e
RuntimeError: Error compiling objects for extension
```

I tried running ```ninja -v``` manually and got the following error:
```
(droidsplat) hygx@hygx:~/code/DROID-Splat$ ninja -v
ninja: error: loading 'build.ninja': No such file or directory
```

any suggestions ?

thanks for  your relply.

JV-X, 2024-12-02:
I switched to the `thirdparty/diff-gaussian-rasterization/diff-gaussian-rasterization-light/build/temp.linux-x86_64-cpython-37` folder, executed `ninja -v`, and got the following output
```
(droidsplat) hygx@hygx:~/code/DROID-Splat/thirdparty/diff-gaussian-rasterization/diff-gaussian-rasterization-light/build/temp.linux-x86_64-cpython-37$ ninja -v
ninja: error: '/mnt/data2/hujiarui/Code/accelerated_gau_slam/submodules/diff-gaussian-rasterization/cuda_rasterizer/backward.cu', needed by '/mnt/data2/hujiarui/Code/accelerated_gau_slam/submodules/diff-gaussian-rasterization/build/temp.linux-x86_64-cpython-37/cuda_rasterizer/backward.o', missing and no known rule to make it
```

ChenHoy, 2024-12-02:
Hi there! 
I sadly do not have much experience with some of the problems you might run into when using windows or WSL. That being said, 
you seem to already have problems with building the SLAM modules. 

Can you post what gcc or g++ compiler version you are using here? Then what CUDA and PyTorch version did you use? 

PS: I dont understand your second post. Our vanilla 3DGS version uses thirdparty/diff-gaussian-rasterization-w-pose as a dependency, what switch did you do? Your build also does not seem to fail because of the splatting modules.

JV-X, 2024-12-02:
thanks for your reply.

my gcc version: 9.4.0
CUDA version: 12.1
PyTorch version：2.1.2

This is the output I got from executing the following commands:

`gcc -v`:

```
(droidsplat) hygx@hygx:~/code/DROID-Splat$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/9/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:hsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 9.4.0-1ubuntu1~20.04.2' --with-bugurl=file:///usr/share/doc/gcc-9/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,gm2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-9 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-9-9QDOt0/gcc-9-9.4.0/debian/tmp-nvptx/usr,hsa --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.2)
```
`g++ -v`:
```
(droidsplat) hygx@hygx:~/code/DROID-Splat$ g++ -v
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/9/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:hsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 9.4.0-1ubuntu1~20.04.2' --with-bugurl=file:///usr/share/doc/gcc-9/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,gm2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-9 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-9-9QDOt0/gcc-9-9.4.0/debian/tmp-nvptx/usr,hsa --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix
gcc version 9.4.0 (Ubuntu 9.4.0-1ubuntu1~20.04.2)
```

`python -c "import torch; print(torch.__version__)"`:
```
2.1.2
```
`nvcc -V`:

```
(droidsplat) hygx@hygx:~/code/DROID-Splat$ nvcc -V
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2023 NVIDIA Corporation
Built on Tue_Feb__7_19:32:13_PST_2023
Cuda compilation tools, release 12.1, V12.1.66
Build cuda_12.1.r12.1/compiler.32415258_0
```

Regarding the second point, I saw an error in the error log when executing `ninja -v`, so I tried to switch to the folder where the "build.ninja" file existed and manually execute the command `ninja -v` to see if an error would be reported.

ChenHoy, 2024-12-02:
Can you try to update your _gcc_? We at least know that the build works with _version 12.3.0_. It would be good to know, if the error persists then. 

Note that with your CUDA version, you might need to change a file in your _pytorch_ package, because of a problem with _nvcc_. See the comment in our README build section: 
You might need to change 45 of `torch/include/pybind11/cast.h`: 
```
-    return caster.operator typename make_caster<T>::template cast_op_type<T>();
+    return caster;
```
Alternatively you could of course switch to a higher version of CUDA

JV-X, 2024-12-03:
thanks， update gcc is work！

ChenHoy, 2024-12-03:
Glad, that this helped you! Closing this issue now. Thank you for your patience :)

closed 2024-12-03 by ChenHoy