ChenHoy/DROID-Splat #9: What should data/Replica.base be?

JV-X, 2024-12-03:
I installed the environment and followed the command in the document to execute `python run.py data=Replica.base data.input_folder=data/ tracking=base stride=1 mode=rgbd`, and then got the following error：

```
(droidsplat) hygx@hygx:~/code/DROID-Splat$ python run.py data=Replica.base data.input_folder=data/ tracking=base stride=
1 mode=rgbd

A module that was compiled using NumPy 1.x cannot be run in
NumPy 2.1.3 as it may crash. To support both 1.x and 2.x
versions of NumPy, modules must be compiled with NumPy 2.0.
Some module may need to rebuild instead e.g. with 'pybind11>=2.12'.

If you are a user of the module, the easiest solution will be to
downgrade to 'numpy<2' or try to upgrade the affected module.
We expect that some modules will need time to support NumPy 2.

Traceback (most recent call last):  File "/home/hygx/code/DROID-Splat/run.py", line 12, in <module>
    import torch
  File "/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/__init__.py", line 1382, in <module>
    from .functional import *  # noqa: F403
  File "/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/functional.py", line 7, in <module>
    import torch.nn.functional as F
  File "/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/nn/__init__.py", line 1, in <module>
    from .modules import *  # noqa: F403
  File "/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/nn/modules/__init__.py", line 35, in <module>
    from .transformer import TransformerEncoder, TransformerDecoder, \
  File "/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/nn/modules/transformer.py", line 20, in <module>
    device: torch.device = torch.device(torch._C._get_default_device()),  # torch.device('cpu'),
/home/hygx/anaconda3/envs/droidsplat/lib/python3.11/site-packages/torch/nn/modules/transformer.py:20: UserWarning: Failed to initialize NumPy: _ARRAY_API not found (Triggered internally at /opt/conda/conda-bld/pytorch_1702400430266/work/torch/csrc/utils/tensor_numpy.cpp:84.)
  device: torch.device = torch.device(torch._C._get_default_device()),  # torch.device('cpu'),
In 'slam': Could not find 'data/Replica.base'

Config search path:
        provider=hydra, path=pkg://hydra.conf
        provider=main, path=file:///home/hygx/code/DROID-Splat/configs
        provider=schema, path=structured://

Set the environment variable HYDRA_FULL_ERROR=1 for a complete stack trace.
(droidsplat) hygx@hygx:~/code/DROID-Splat$
```

What should I modify to make the program predict the correct results?

ChenHoy, 2024-12-03:
Hi, so as you can see in file `configs/data/Replica/base.yaml`, the `input_folder` needs to point to a specific scene. For example you would use it like this:  
`python run.py data=Replica.base data.input_folder=data/Replica/office0/ tracking=base stride=1 mode=rgbd`

Hope, that this helps you!

JV-X, 2024-12-05:
> Hi, so as you can see in file `configs/data/Replica/base.yaml`, the `input_folder` needs to point to a specific scene. For example you would use it like this: `python run.py data=Replica.base data.input_folder=data/Replica/office0/ tracking=base stride=1 mode=rgbd`
> 
> Hope, that this helps you!

thanks for your  reply！

I don't think I understand what you mean. I tried changing the `input_folder` in `base.yaml` to `/home/hygx/code/DROID-Splat/data`, where data contains 10 photos. Then I tried both
 `python run.py data=Replica.base data.input_folder=data/ tracking=base stride=1 mode=rgbd` 
and
 `python run.py data=Replica.base data.input_folder=data/Replica/office0/ tracking=base stride=1 mode=rgbd`, 

but got the same error.

ChenHoy, 2024-12-05:
Hi,
I am sorry if this was confusing. So the way it works, and you can see that in the scripts at `scripts/evaluation/evaluate_on_replica.sh`, is you use: 
`python run.py data=Replica/base data.input_folder=/data/Replica/office0/ tracking=base mapping=replica`. I think your command did not work, because you used `data=Replica.base`.

If you have your own dataset or want to run it on a random folder, then we assume the following structure: 

/path/to/your/dataset
├── images
│   ├── frame_0001.png
│   ├── frame_0002.png
│   ├── frame_0003.png
|   ...
│   └── frame_.0010.png
├── metric3d-vit_giant2
│   ├── frame_0001.npy
│   ├── frame_0002.npy
│   ├── frame_0003.npy
|   ...
│   └── frame_0010.npy
├── depth
│   ├── frame_0001.png
│   ├── frame_0002.png
│   ├── frame_0003.png
|   ...
│   └── frame_0010.png

The logic behind this decision is that we might supply more than just images to the system, i.e. monocular depth, intrinsics, depth or masks. So we always assume you set the custom input_folder to `/path/to/your/folder/`, where the actual images are in `/path/to/your/folder/images`. 

I hope this helps

Emmanuel9012, 2024-12-24:
@JV-X make sure you override the default config.(data=Replica/base.yamll)

JV-X, 2024-12-25:
> @JV-X make sure you override the default config.(data=Replica/base.yamll)

sorry，I'm still a little bit confused, can you be more specific? What field in what file should I change to what value?

ChenHoy, 2024-12-25:
Hi, sorry for the late reply. 

So hydra works in a way, that you can either change the default file path at `configs/data/Replica/base.yaml` under  the field `input_folder` **or** you can just override from the command line by e.g. using `python run.py data.input_folder=path/to/your/scene`. Its important that this would be the actual scene folder and not just dataset base path, i.e. for Replica you would for example choose `data.input_folder=/path/to/Replica/room0`.

ChenHoy, 2025-01-02:
I assume that this got resolved due to inactivity. Please reopen if needed :)

closed 2025-01-02 by ChenHoy