ChenHoy/DROID-Splat #11: No module named 'src.gaussian_splatting.utils.reloc_utils'

Andymulb, 2024-12-07:
Hi, thank you for your awesome work!

I just tried to install DROID-Splat on an NVIDIA Jetson Orin with CUDA 12.6 and a specific PyTorch Version (2.5.0a0+872d972e41.nv24.08). The installation (mcmc branch) did complete successfully. However, I needed to build PyOpenGL-accelerate from source and conda install did not work for many dependencies, so I used pip instead.

When trying to run the basic inference command "python run.py data=Replica.base data.input_folder=/media/data/Replica/office0 tracking=base stride=1 mode=rgbd" I'm receiving the following error message:
python run.py data=Replica.base data.input_folder=/media/data/Replica/office0 tracking=base stride=1 mode=rgbd
/home/DROID-Splat/src/factor_graph.py:106: FutureWarning: `torch.cuda.amp.autocast(args...)` is deprecated. Please use `torch.amp.autocast('cuda', args...)` instead.
  @torch.cuda.amp.autocast(enabled=True)
/home/DROID-Splat/src/factor_graph.py:162: FutureWarning: `torch.cuda.amp.autocast(args...)` is deprecated. Please use `torch.amp.autocast('cuda', args...)` instead.
  @torch.cuda.amp.autocast(enabled=True)
/home/DROID-Splat/src/factor_graph.py:189: FutureWarning: `torch.cuda.amp.autocast(args...)` is deprecated. Please use `torch.amp.autocast('cuda', args...)` instead.
  @torch.cuda.amp.autocast(enabled=True)
/home/DROID-Splat/src/factor_graph.py:444: FutureWarning: `torch.cuda.amp.autocast(args...)` is deprecated. Please use `torch.amp.autocast('cuda', args...)` instead.
  @torch.cuda.amp.autocast(enabled=True)
/home/DROID-Splat/src/factor_graph.py:512: FutureWarning: `torch.cuda.amp.autocast(args...)` is deprecated. Please use `torch.amp.autocast('cuda', args...)` instead.
  @torch.cuda.amp.autocast(enabled=False)
/home/DROID-Splat/src/factor_graph.py:588: FutureWarning: `torch.cuda.amp.autocast(args...)` is deprecated. Please use `torch.amp.autocast('cuda', args...)` instead.
  @torch.cuda.amp.autocast(enabled=True)
Traceback (most recent call last):
  File "/home/DROID-Splat/run.py", line 14, in <module>
    from src.slam import SLAM
  File "/home/DROID-Splat/src/slam.py", line 22, in <module>
    from .frontend import FrontendWrapper
  File "/home/DROID-Splat/src/frontend.py", line 11, in <module>
    from .motion_filter import MotionFilter
  File "/home/DROID-Splat/src/motion_filter.py", line 8, in <module>
    from .depth_video import DepthVideo
  File "/home/DROID-Splat/src/depth_video.py", line 17, in <module>
    from .gaussian_splatting.gaussian_renderer import render
  File "/home/DROID-Splat/src/gaussian_splatting/gaussian_renderer/__init__.py", line 19, in <module>
    from ..scene.gaussian_model import GaussianModel
  File "/home/DROID-Splat/src/gaussian_splatting/scene/gaussian_model.py", line 37, in <module>
    from ..utils.reloc_utils import compute_relocation_cuda
ModuleNotFoundError: No module named 'src.gaussian_splatting.utils.reloc_utils'

Is this a known issue? I couldn't find the missing file in your repo.

ChenHoy, 2024-12-07:
Hi, thank you for reporting this. The file was indeed missing. 

Pushed a fix, does it work now for you?

ChenHoy, 2024-12-08:
I just noticed, you also said that the conda build did not work for you. Which packages did you need to install via pip? I only observed this problem with _open3d_ on my environment. Would appreciate info on this ...

Andymulb, 2024-12-09:
Thank you! The fix helped, but then I got another error message on the Jetson:

[Main]: Load pretrained checkpoint from ./pretrained/droid.pth!
/home/DROID-Splat/src/slam.py:260: FutureWarning: You are using `torch.load` with `weights_only=False` (the current default value), which uses the default pickle module implicitly. It is possible to construct malicious pickle data which will execute arbitrary code during unpickling (See https://github.com/pytorch/pytorch/blob/main/SECURITY.md#untrusted-models for more details). In a future release, the default value for `weights_only` will be flipped to `True`. This limits the functions that could be executed during unpickling. Arbitrary objects will no longer be allowed to be loaded via this mode unless they are explicitly allowlisted by the user via `torch.serialization.add_safe_globals`. We recommend you start setting `weights_only=True` for any use case where you don't have full control of the loaded file. Please open an issue on GitHub for any issues related to this experimental feature.
  state_dict = OrderedDict([(k.replace("module.", ""), v) for (k, v) in torch.load(pretrained).items()])
[2024-12-09 08:08:06,685][__main__][INFO] - Running on 2000 frames
/home/anaconda3/envs/DROID-Splat/lib/python3.10/site-packages/torchvision/io/image.py:13: UserWarning: Failed to load image Python extension: 
  warn(f"Failed to load image Python extension: {e}")
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/anaconda3/envs/DROID-Splat/lib/python3.10/multiprocessing/spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "/home/anaconda3/envs/DROID-Splat/lib/python3.10/multiprocessing/spawn.py", line 126, in _main
    self = reduction.pickle.load(from_parent)
  File "/home/anaconda3/envs/DROID-Splat/lib/python3.10/site-packages/torch/multiprocessing/reductions.py", line 149, in rebuild_cuda_tensor
    storage = storage_cls._new_shared_cuda(
  File "/home/anaconda3/envs/DROID-Splat/lib/python3.10/site-packages/torch/storage.py", line 1420, in _new_shared_cuda
    return torch.UntypedStorage._new_shared_cuda(*args, **kwargs)
RuntimeError: CUDA error: invalid argument
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.

I also just tested DROID-Splat on a different x86-based system with an RTX 4080 GPU, and everything works fine. However, the Jetson AGX Orin has 64 GB of unified memory. Therefore, it shouldn't be a memory-related issue. Do you have another idea what might be the problem?

The conda build worked instantly on the x86-based system. The Jetson is an aarch64-based platform, and conda often does not provide packages for it. I needed to install the following alternative packages:

- pytorch_scatter → torch-scatter (pip)
- cudatoolkit → cuda-toolkit=12.6 (conda-forge)
- opencv → opencv-python (pip)

Since there were more dependency conflicts, I tried to install as many packages as possible via pip. I guess the only one I installed via conda (apart from cuda-toolkit) is suitesparse.

ChenHoy, 2024-12-09:
Hi, glad that the MCMC branch now works. 

Your information on the build process on Jetson GPU's seems very valuable, I am sure this will be helpful to other users as well.

As I don't have much experience with these systems, I am not sure if I can help you. 
From your log, it seems like `torch.multiprocessing` has some pitfalls on that specific architecture. My first idea would be to try out different pytorch and CUDA versions. The error message is kind of cryptic, it seems related to torch's shared memory space, which we use in the `depth_video` buffer tensors in order to access the same tensors in different Processes. Therefore, I would search for related issues for `torch.tensor.share_memory_()`  for your GPU.

ChenHoy, 2024-12-10:
Closing this issue, as the original problem of the missing file has been resolved. I opened a new one for the problems on Jetson GPU's.

closed 2024-12-10 by ChenHoy