ChenHoy/DROID-Splat #12: OOM Error

Tokyo-dream, 2024-12-08:
Thanks for this work！What are the CUDA memory requirements? My device is V100 32G, whether it is to run TUM or Replica data set, it will be OOM
================================================================
  File "DROID-Splat/src/losses/image.py", line 212, in _ssim
    cs_map = (2 * sigma12 + C2) / (sigma1_sq + sigma2_sq + C2)  # set alpha=beta=gamma=1
                                   ~~~~~~~~~~~~~~~~~~~~~~^~~~
torch.cuda.OutOfMemoryError: CUDA out of memory. Tried to allocate 20.00 MiB. GPU 0 has a total capacty of 31.74 GiB of which 10.38 MiB is free. Process 47923 has 2.20 GiB memory in use. Process 48771 has 29.22 GiB memory in use. Process 48851 has 306.00 MiB memory in use. Of the allocated memory 27.27 GiB is allocated by PyTorch, and 1.37 GiB is reserved by PyTorch but unallocated. If reserved but unallocated memory is large try setting max_split_size_mb to avoid fragmentation.  See documentation for Memory Management and PYTORCH_CUDA_ALLOC_CONF

Tokyo-dream, 2024-12-08:
[image]
like thithis

closed 2024-12-08 by Tokyo-dream

ChenHoy, 2024-12-08:
Hey there! Just seeing this closed issue. What was the reason for this problem and how did it get resolved?

We used a RTX 4090 with 24GB in all of our experiments. What data and with what configs did you try this on? 

Since you get the OOM during refinement, it seems likely that you have accumulated a lot of Gaussians > 500k. We observed, that in many scenes you can even get away with using a million Gaussians depending on how you optimize. That being said, the configs for replica and tum in `conigs/mapping/` should never use > 24GB memory.