ChenHoy/DROID-Splat #13: NVIDIA Jetson GPU's: RuntimeError: CUDA error: invalid argument in shared_memory

ChenHoy, 2024-12-10:
Thank you! The fix helped, but then I got another error message on the Jetson:

[Main]: Load pretrained checkpoint from ./pretrained/droid.pth!
/home/DROID-Splat/src/slam.py:260: FutureWarning: You are using `torch.load` with `weights_only=False` (the current default value), which uses the default pickle module implicitly. It is possible to construct malicious pickle data which will execute arbitrary code during unpickling (See https://github.com/pytorch/pytorch/blob/main/SECURITY.md#untrusted-models for more details). In a future release, the default value for `weights_only` will be flipped to `True`. This limits the functions that could be executed during unpickling. Arbitrary objects will no longer be allowed to be loaded via this mode unless they are explicitly allowlisted by the user via `torch.serialization.add_safe_globals`. We recommend you start setting `weights_only=True` for any use case where you don't have full control of the loaded file. Please open an issue on GitHub for any issues related to this experimental feature.
  state_dict = OrderedDict([(k.replace("module.", ""), v) for (k, v) in torch.load(pretrained).items()])
[2024-12-09 08:08:06,685][__main__][INFO] - Running on 2000 frames
/home/anaconda3/envs/DROID-Splat/lib/python3.10/site-packages/torchvision/io/image.py:13: UserWarning: Failed to load image Python extension: 
  warn(f"Failed to load image Python extension: {e}")
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/anaconda3/envs/DROID-Splat/lib/python3.10/multiprocessing/spawn.py", line 116, in spawn_main
    exitcode = _main(fd, parent_sentinel)
  File "/home/anaconda3/envs/DROID-Splat/lib/python3.10/multiprocessing/spawn.py", line 126, in _main
    self = reduction.pickle.load(from_parent)
  File "/home/anaconda3/envs/DROID-Splat/lib/python3.10/site-packages/torch/multiprocessing/reductions.py", line 149, in rebuild_cuda_tensor
    storage = storage_cls._new_shared_cuda(
  File "/home/anaconda3/envs/DROID-Splat/lib/python3.10/site-packages/torch/storage.py", line 1420, in _new_shared_cuda
    return torch.UntypedStorage._new_shared_cuda(*args, **kwargs)
RuntimeError: CUDA error: invalid argument
Compile with `TORCH_USE_CUDA_DSA` to enable device-side assertions.

I also just tested DROID-Splat on a different x86-based system with an RTX 4080 GPU, and everything works fine. However, the Jetson AGX Orin has 64 GB of unified memory. Therefore, it shouldn't be a memory-related issue. Do you have another idea what might be the problem?

The conda build worked instantly on the x86-based system. The Jetson is an aarch64-based platform, and conda often does not provide packages for it. I needed to install the following alternative packages:

- pytorch_scatter → torch-scatter (pip)
- cudatoolkit → cuda-toolkit=12.6 (conda-forge)
- opencv → opencv-python (pip)

Since there were more dependency conflicts, I tried to install as many packages as possible via pip. I guess the only one I installed via conda (apart from cuda-toolkit) is suitesparse.

_Originally posted by @Andymulb in https://github.com/ChenHoy/DROID-Splat/issues/11#issuecomment-2527860575_

Andymulb, 2025-01-09:
I was able to solve the issue by simply compiling the code with "-gencode=arch=compute_87,code=sm_87", in the setup.py file of DROID-Splat and all third-party components

ChenHoy, 2025-01-10:
Ok, very nice! Closing this issue then as its resolved

closed 2025-01-10 by ChenHoy