ChenHoy/DROID-Splat #14: Deadlock on gaussian optimization

LaFeuilleMorte, 2024-12-11:
When I run the run.py with a bunch of images, the process will get stuck at a certain step, After waiting for a very long time, I pressed the Ctrl+C to terminate the process. And I received the following log. It seemed that it has been stuck on the global_ba at here: semaBackend.acquire() .

Process Frontend Tracking:
Traceback (most recent call last):
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/aistudio/workspace/aigc/wangqihang013/aigc3d/repos/neural_rendering/sfm/DROID-Splat/src/slam.py", line 434, in global_ba
    semaBackend.acquire()  # Aquire the semaphore (If the counter == 0, then this thread will be blocked)
KeyboardInterrupt
Traceback (most recent call last):
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/aistudio/workspace/aigc/wangqihang013/aigc3d/repos/neural_rendering/sfm/DROID-Splat/src/slam.py", line 308, in tracking
    condTracking.wait_for(lambda: self.mapping_done > 0)
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/multiprocessing/synchronize.py", line 313, in wait_for
    self.wait(waittime)
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/multiprocessing/synchronize.py", line 261, in wait
    return self._wait_semaphore.acquire(True, timeout)
KeyboardInterrupt
Traceback (most recent call last):
  File "/aistudio/workspace/aigc/wangqihang013/aigc3d/repos/neural_rendering/sfm/DROID-Splat/run.py", line 119, in <module>
    run_slam()
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/site-packages/hydra/main.py", line 94, in decorated_main
    _run_hydra(
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/site-packages/hydra/_internal/utils.py", line 394, in _run_hydra
    _run_app(
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/site-packages/hydra/_internal/utils.py", line 457, in _run_app
    run_and_report(
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/site-packages/hydra/_internal/utils.py", line 220, in run_and_report
    return func()
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/site-packages/hydra/_internal/utils.py", line 458, in <lambda>
    lambda: hydra.run(
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/site-packages/hydra/_internal/hydra.py", line 119, in run
    ret = run_job(
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/site-packages/hydra/core/utils.py", line 186, in run_job
    ret.return_value = task_function(task_cfg)
  File "/aistudio/workspace/aigc/wangqihang013/aigc3d/repos/neural_rendering/sfm/DROID-Splat/run.py", line 114, in run_slam
    slam.run(dataset)
  File "/aistudio/workspace/aigc/wangqihang013/aigc3d/repos/neural_rendering/sfm/DROID-Splat/src/slam.py", line 1074, in run
    while self.mapping_queue.empty():
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/multiprocessing/queues.py", line 129, in empty
    return not self._poll()
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/multiprocessing/connection.py", line 257, in poll
    return self._poll(timeout)
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/multiprocessing/connection.py", line 424, in _poll
    r = wait([self], timeout)
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/multiprocessing/connection.py", line 925, in wait
    selector.register(obj, selectors.EVENT_READ)
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/selectors.py", line 353, in register
    key = super().register(fileobj, events, data)
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/selectors.py", line 239, in register
    key = SelectorKey(fileobj, self._fileobj_lookup(fileobj), events, data)
  File "<string>", line 1, in <lambda>
KeyboardInterrupt
Traceback (most recent call last):
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/aistudio/workspace/aigc/wangqihang013/aigc3d/repos/neural_rendering/sfm/DROID-Splat/src/slam.py", line 559, in gaussian_mapping
    self.gaussian_mapper(mapping_queue, received_mapping)
  File "/aistudio/workspace/aigc/wangqihang013/aigc3d/repos/neural_rendering/sfm/DROID-Splat/src/gaussian_mapping.py", line 1119, in __call__
    self._update(iters=self.mapping_iters)
  File "/aistudio/workspace/aigc/wangqihang013/aigc3d/repos/neural_rendering/sfm/DROID-Splat/src/gaussian_mapping.py", line 1070, in _update
    loss = self.mapping_step(
  File "/aistudio/workspace/aigc/wangqihang013/aigc3d/repos/neural_rendering/sfm/DROID-Splat/src/gaussian_mapping.py", line 707, in mapping_step
    current_loss, render_pkg = self.render_compare(view)
  File "/aistudio/workspace/aigc/wangqihang013/aigc3d/repos/neural_rendering/sfm/DROID-Splat/src/gaussian_mapping.py", line 685, in render_compare
    current_loss = mapping_rgbd_loss(image, depth, view, **self.loss_params)
  File "/aistudio/workspace/aigc/wangqihang013/aigc3d/repos/neural_rendering/sfm/DROID-Splat/src/losses/__init__.py", line 61, in mapping_rgbd_loss
    loss_depth = depth_loss(
  File "/aistudio/workspace/aigc/wangqihang013/aigc3d/repos/neural_rendering/sfm/DROID-Splat/src/losses/depth.py", line 54, in depth_loss
    depth_loss += beta * edge_weighted_tv(depth_est, original_image, mask=mask)
RuntimeError: a leaf Variable that requires grad is being used in an in-place operation.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/multiprocessing/process.py", line 317, in _bootstrap
    util._exit_function()
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/multiprocessing/util.py", line 360, in _exit_function
    _run_finalizers()
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/multiprocessing/util.py", line 300, in _run_finalizers
    finalizer()
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/multiprocessing/util.py", line 224, in __call__
    res = self._callback(*self._args, **self._kwargs)
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/multiprocessing/queues.py", line 199, in _finalize_join
    thread.join()
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/threading.py", line 1096, in join
    self._wait_for_tstate_lock()
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/threading.py", line 1116, in _wait_for_tstate_lock
    if lock.acquire(block, timeout):
KeyboardInterrupt
^CException ignored in atexit callback: <function _exit_function at 0x7fc4ee2003a0>
Traceback (most recent call last):
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/multiprocessing/util.py", line 357, in _exit_function
    p.join()
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/multiprocessing/process.py", line 149, in join
    res = self._popen.wait(timeout)
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/multiprocessing/popen_fork.py", line 43, in wait
    return self.poll(os.WNOHANG if timeout == 0.0 else 0)
  File "/aistudio/workspace/system-default/envs/droidsplat/lib/python3.10/multiprocessing/popen_fork.py", line 27, in poll
    pid, sts = os.waitpid(self.pid, flag)

ChenHoy, 2024-12-11:
Hi, thank you for reporting this. As can be seen, there was a bug when we refactored the losses. 

Since you used an additional smoothness term for the depth loss, we performed   
`depth_loss += beta * edge_weighted_tv(depth_est, original_image, mask=mask)`
, which logically raised _RuntimeError: a leaf Variable that requires grad is being used in an in-place operation_.

Fixed this on the latest push by not doing an in-place add. Can you confirm, that this got resolved?

ChenHoy, 2024-12-11:
Can you provide more details on which Processes you finally run? 

496k Gaussians can be quite a lot, depending on the scene and if you run all other modules in parallel at a higher enough resolution, things add up. I can also recommend, that sometimes less Gaussians provide better performance, we usually aimed for 100 - 250k Gaussians on our scenes. 

see https://github.com/ChenHoy/DROID-Splat/issues/11#issuecomment-2525268441 for some tips on how to reduce the memory footprint of the system. If you are truly determined to have a lot of Gaussians, its smart to run the system on a low resoution and with few Gaussians during online mode and then densify more aggressively during refinement when other Processes are already shutdown. This would require tuning `mapping.refinement.prune_densify_every`, `mapping.refinement.densify_until` and the parameters of `mapping.refinement.densify`.

ChenHoy, 2024-12-11:
Closing this issue, since the deadlock got resolved. If you still observe unpredictable OOM's or another bug, feel free to open a new issue :) 

Cheers
Christian

closed 2024-12-11 by ChenHoy