ChenHoy/DROID-Splat #16: I tried on Cuda 11.7 with Pytorch 2.0.1 and it is never finishing. It also consumes a very large GPU, like 25 GB. The following is from the terminal after waiting for more than 36 hours

Emmanuel9012, 2024-12-14:
497/500 [06:47<00:02,  1.18it/s[Gaussian Mapper]  Refinement loss: 0.032515283674001694                                                                                                                                 | 0/1 [00:00<?, ?it/s]
Batch Optimization: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:00<00:00,  1.22it/s]
Gaussian Refinement: 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████▍| 498/500 [06:48<00:01,  1.20it/s[Gaussian Mapper]  Refinement loss: 0.032300081104040146                                                                                                                                 | 0/1 [00:00<?, ?it/s]
Batch Optimization: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:00<00:00,  1.25it/s]
Gaussian Refinement: 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████▋| 499/500 [06:49<00:00,  1.21it/s[Gaussian Mapper]  Refinement loss: 0.03311951085925102                                                                                                                                  | 0/1 [00:00<?, ?it/s]
Batch Optimization: 100%|███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:00<00:00,  1.22it/s]
Gaussian Refinement: 100%|██████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 500/500 [06:49<00:00,  1.22it/s]
[Gaussian Mapper] Gaussians after Map Refinement: 170271
[Gaussian Mapper] Mapping refinement finished
[Gaussian Mapper] Mesh saved at /mnt/data/Students/Aman/DROID-Splat/outputs/droidsplat/2024-12-13_03-20-24_rgbd/mesh/final_rgbd.ply
[Gaussian Mapper]  Final mapping loss: 0.03311951085925102
[Gaussian Mapper] 11 iterations, 6.7272727272727275 cams/it
[2024-12-13 03:46:22,079][src.slam][INFO] - [Main]: Received final mapping update!

ChenHoy, 2024-12-14:
Hi there, 
so from what I can see the Processes actually finished. I assume you use this with GUI enabled? You have to manually close those windows in order to terminate properly. If you want to evaluate, I would recommend to just run without GUI. 

W.r.t the memory footprint: What resolution are you running on and what other configurations? Since you dont seem to have a lot of Gaussians, I guess most of this comes from the Tracking system. If this is critical, then you might need to change some of the parameters of the backend or run on a lower resolution

ChenHoy, 2025-01-02:
From your reaction, I take it that this got resolved? Closing this issue now, please reopen if this needs further attention

closed 2025-01-02 by ChenHoy