ChenHoy/DROID-Splat #17: How to operate without providing camera intrinsic parameters?

ScottishOwl, 2025-02-15:
Hi, I want to study the impact of camera intrinsics on PSNR and observe how accurately the model predicts camera intrinsic parameters. What configuration settings are required to enable operation without camera intrinsics?

ChenHoy, 2025-02-19:
Hi, sorry for the late reply. 

So this is actually a very interesting question which we did not evaluate quantitatively too much in our paper. The original intrinsics optimization [paper](https://openaccess.thecvf.com/content/ICCV2023/papers/Hagemann_Deep_Geometry-Aware_Camera_Self-Calibration_from_Video_ICCV_2023_paper.pdf) provides answers for this in terms of VO quality against initial calibration errors. 

In order to run the intrinsics optimization, use: 
`python run.py opt_intr=True`

This will use an optimization kernel, which **optimizes camera poses, disparity and intrinsics** during Bundle Adjustment. Its pretty clear, that in case you do not have intrinsics or bad ones, you will likely get better results and correct the intrinsics parameters (This still depends on the camera odometry, i.e. if you move in a straight line, this will not be helpful for optimizing intrinsics) with this optimization enabled. This is normally pretty robust. We observed this qualitatively in our experiments as well, i.e. for many hand-captured scenes you get less distorted trajectories and 3D structure.  

The reason, we have not evaluated this quantitatively for Rendering, is that I believe that you can still overfit Gaussians to get good Renderings metrics even with bad intrinsics. The gap between _keyframe (train)_ and _non-keyframe (test)_ images will likely just widen a little bit (right now its usually ~1-2 PSNR). 
I think it **makes more sense to evaluate on actual 3D reconstruction metrics**, however these only exist on selected datasets.

**How do we start with no intrinsics?**
Let's assume you run this on a self-captured video and you already have our datastructure, i.e. a folder with subfolders `my_video/images/` and maybe `my_video/mono_depth_model/` and in these are the respective frames. You would want to have a _.yaml_ config file for this in `configs/data/Custom/my_new_video.yaml`. 
The intrinsics of a dataset are specified in those files. You could also overwrite them manually (hydra expects null instead of None) to just simply be None: 

`python run.py data.cam.fx=null data.cam.fy=null data.cam.cx=null data.cam.cy=null opt_intr=True`

You can see that in our main script `run.py`, we check for missing intrinsics and use a heuristic for the initial value dependent on the image size. 

**What do we have to be careful about?**
One problem with quantitative evaluation on casual video, is that we do not have metric depth. However, we find that running with monocular depth priors helps a lot compared to pure monocular video. 
Now, you cannot run both scale-optimization (tracking.frontend.optimize_scales=True) and intrinsics optimization due to a scale ambiguity. If you consider our workflow in the official _README.md_, you would end up with a consistent map and fitting intrinsics, but the scale will not be correct. 
I think for this reason, it makes sense if you either use a monocular prior and manually align it to a reference or run with scale-correct depth priors and make a sensitivity analysis of the intrinsics.

I hope this answer helps you and I would be interested in how you progress on this. 

Best regards
Christian

ScottishOwl, 2025-02-21:
Thanks for your reply! I'll have a try under this guidance some other day. Anyway, your work is quite extraordinary.
Wish you a good day!

ChenHoy, 2025-03-06:
Closing for the time being, please reopen when needed

closed 2025-03-06 by ChenHoy