ChenHoy/DROID-Splat #19: for replica

galahies, 2025-03-01:
The replica dataset I downloaded was lost due to my mistake ，When I download it again for this link:wget https://cvg-data.inf.ethz.ch/nice-slam/data/Replica.zip （only for 11.59GB）
There is only the result part in the dataset, and the RGB image part of colors is lost
I don't know if you have any copies of the colors, but I would appreciate it if you could provide it to me

ChenHoy, 2025-03-06:
Hi, you could also download from the [GO-Slam](https://github.com/youmi-zym/GO-SLAM) repo with this  [download link](https://drive.google.com/drive/folders/1RJr38jvmuIV717PCEcBkzV2qkqUua-Fx?usp=sharing)
, which lets you download from Google Drive. Hope this helps in case the original dataset is not up again :)

zyh778, 2025-03-16:
I have the same problem, have you found this dataset yet?

> The replica dataset I downloaded was lost due to my mistake ，When I download it again for this link:wget https://cvg-data.inf.ethz.ch/nice-slam/data/Replica.zip （only for 11.59GB） There is only the [...]

zyh778, 2025-03-16:
> Hi, you could also download from the [GO-Slam](https://github.com/youmi-zym/GO-SLAM) repo with this [download [...]

There's no more RGB datasets here either

ChenHoy, 2025-03-21:
Oh wow, you are right. How about going directly to the source at [Meta](https://github.com/facebookresearch/Replica-Dataset)?

zyh778, 2025-03-21:
However, if I directly use the original dataset or convert it myself to formats like RGB, is it likely to cause unnecessary errors? Could you please share the dataset you used, perhaps via a Google Drive link or another method? I would be extremely grateful.

> Oh wow, you are right. How about going directly to the source at [Meta](https://github.com/facebookresearch/Replica-Dataset)?

ChenHoy, 2025-04-16:
I will close this issue now as it got resolved with my upload. If anything is needed, feel free to reopen :)

closed 2025-04-16 by ChenHoy