ChenHoy/DROID-Splat #20: experimental results

shezhi, 2025-03-06:
Hello Author, thank you very much for sharing the experiment. I would like to ask you a question. After downloading the code and manually configuring the environment, I ran the command:
**python run.py data=TUM_RGBD/fr1 stride=1 data.input_folder=/home/si/rgbd_dataset_freiburg1_xyz tracking=tum mapping=tum backend_every=8 mapper_every=20 mono=rgbd** 
but there were no parameters shown and the result is as follows. What could be the reason for this? I look forward to your reply. 

[image]

shezhi, 2025-03-06:
No parameters were modified.

ChenHoy, 2025-03-06:
Hi, that looks super odd. You can at least see on the right of the GUI, that the correct input RGBD frames are feeded into the Renderer/Mapper. I think your command is wrong, i.e. you used `mono=rgbd` instead of `mode=rgbd`. So I suspect that something is messed up because of that. 

Further comment: is _xyz_ not a scene from the freiburg2 sequence? I thought it needs the fr2.yaml for calibration ... 

If this not helps, can you provide maybe further information on the full config with which this was run?

shezhi, 2025-03-06:
已收到！

shezhi, 2025-03-06:
Hello, thank you for your reply. The issue has been resolved, and the reason was my carelessness! I forgot to execute that command.
**python setup.py install**
 I apologize for the inconvenience, and I really appreciate your response and the work you shared.

closed 2025-03-06 by shezhi

ChenHoy, 2025-03-21:
Thank you, glad that everything works out!