ChenHoy/DROID-Splat #25: Gaussian refinement freezing issue

zyh778, 2025-03-29:
When I followed the complete configuration steps according to the README and executed the command:
```
git clone --recursive https://github.com/ChenHoy/DROID-Splat.git
cd DROID-Splat

sudo apt-get install libopenexr-dev
    
conda env create -f environment.yaml
conda activate droidsplat

python setup.py install

python run.py data=Replic/base data.input_folder=/media/data/Replica/office0 tracking=base stride=1 mode=rgbd

```
Moreover, no other parameters were changed.the program froze completely after running for a while and stopped responding. 

[image]

After pressing Ctrl+C to terminate it, the following error was reported:

```
Traceback (most recent call last):
  File "/home/zyh/DROID-Splat/run.py", line 119, in <module>
Process Gaussian Mapping:
    run_slam()
  File "/home/zyh/miniconda3/envs/DS/lib/python3.11/site-packages/hydra/main.py", line 94, in decorated_main
    _run_hydra(
  File "/home/zyh/miniconda3/envs/DS/lib/python3.11/site-packages/hydra/_internal/utils.py", line 394, in _run_hydra
    _run_app(
  File "/home/zyh/miniconda3/envs/DS/lib/python3.11/site-packages/hydra/_internal/utils.py", line 457, in _run_app
    run_and_report(
  File "/home/zyh/miniconda3/envs/DS/lib/python3.11/site-packages/hydra/_internal/utils.py", line 220, in run_and_report
    return func()
           ^^^^^^
  File "/home/zyh/miniconda3/envs/DS/lib/python3.11/site-packages/hydra/_internal/utils.py", line 458, in <lambda>
    lambda: hydra.run(
            ^^^^^^^^^^
  File "/home/zyh/miniconda3/envs/DS/lib/python3.11/site-packages/hydra/_internal/hydra.py", line 119, in run
    ret = run_job(
          ^^^^^^^^
  File "/home/zyh/miniconda3/envs/DS/lib/python3.11/site-packages/hydra/core/utils.py", line 186, in run_job
    ret.return_value = task_function(task_cfg)
                       ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zyh/DROID-Splat/run.py", line 114, in run_slam
    slam.run(dataset)
  File "/home/zyh/DROID-Splat/src/slam.py", line 1074, in run
    while self.mapping_queue.empty():
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zyh/miniconda3/envs/DS/lib/python3.11/multiprocessing/queues.py", line 129, in empty
    return not self._poll()
               ^^^^^^^^^^^^
  File "/home/zyh/miniconda3/envs/DS/lib/python3.11/multiprocessing/connection.py", line 253, in poll
    def poll(self, timeout=0.0):

KeyboardInterrupt
Traceback (most recent call last):
  File "/home/zyh/miniconda3/envs/DS/lib/python3.11/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/zyh/miniconda3/envs/DS/lib/python3.11/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/zyh/DROID-Splat/src/slam.py", line 580, in gaussian_mapping
    finished = self.gaussian_mapper(mapping_queue, received_mapping, True)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zyh/DROID-Splat/src/gaussian_mapping.py", line 1138, in __call__
    self._last_call(mapping_queue=mapping_queue, received_item=received_item)
  File "/home/zyh/DROID-Splat/src/gaussian_mapping.py", line 930, in _last_call
    self.map_refinement()
  File "/home/zyh/DROID-Splat/src/gaussian_mapping.py", line 540, in map_refinement
    loss = self.mapping_step(
           ^^^^^^^^^^^^^^^^^^
  File "/home/zyh/DROID-Splat/src/gaussian_mapping.py", line 707, in mapping_step
    current_loss, render_pkg = self.render_compare(view)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zyh/DROID-Splat/src/gaussian_mapping.py", line 685, in render_compare
    current_loss = mapping_rgbd_loss(image, depth, view, **self.loss_params)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zyh/DROID-Splat/src/losses/__init__.py", line 61, in mapping_rgbd_loss
    loss_depth = depth_loss(
                 ^^^^^^^^^^^
  File "/home/zyh/DROID-Splat/src/losses/depth.py", line 54, in depth_loss
    depth_loss = depth_loss + beta * edge_weighted_tv(depth_est, original_image, mask=mask)
                                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zyh/DROID-Splat/src/losses/misc.py", line 29, in edge_weighted_tv
    ref_x, ref_y = gradient_map(ref, return_xy=True)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zyh/DROID-Splat/src/utils/image_utils.py", line 56, in gradient_map
    [torch.nn.functional.conv2d(image[i].unsqueeze(0), kernel_x, padding=1) for i in range(image.shape[0])]
  File "/home/zyh/DROID-Splat/src/utils/image_utils.py", line 56, in <listcomp>
    [torch.nn.functional.conv2d(image[i].unsqueeze(0), kernel_x, padding=1) for i in range(image.shape[0])]
     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
torch.cuda.OutOfMemoryError: CUDA out of memory. Tried to allocate 2.00 MiB. GPU 0 has a total capacty of 23.57 GiB of which 5.50 MiB is free. Process 1983121 has 2.60 GiB memory in use. Including non-PyTorch memory, this process has 20.67 GiB memory in use. Process 1984575 has 254.00 MiB memory in use. Of the allocated memory 19.14 GiB is allocated by PyTorch, and 871.62 MiB is reserved by PyTorch but unallocated. If reserved but unallocated memory is large try setting max_split_size_mb to avoid fragmentation.  See documentation for Memory Management and PYTORCH_CUDA_ALLOC_CONF

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/zyh/miniconda3/envs/DS/lib/python3.11/multiprocessing/process.py", line 317, in _bootstrap
    util._exit_function()
  File "/home/zyh/miniconda3/envs/DS/lib/python3.11/multiprocessing/util.py", line 363, in _exit_function
    _run_finalizers()
  File "/home/zyh/miniconda3/envs/DS/lib/python3.11/multiprocessing/util.py", line 303, in _run_finalizers
    finalizer()
  File "/home/zyh/miniconda3/envs/DS/lib/python3.11/multiprocessing/util.py", line 227, in __call__
    res = self._callback(*self._args, **self._kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zyh/miniconda3/envs/DS/lib/python3.11/multiprocessing/queues.py", line 199, in _finalize_join
    thread.join()
  File "/home/zyh/miniconda3/envs/DS/lib/python3.11/threading.py", line 1119, in join
    self._wait_for_tstate_lock()
  File "/home/zyh/miniconda3/envs/DS/lib/python3.11/threading.py", line 1139, in _wait_for_tstate_lock
    if lock.acquire(block, timeout):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^

```

ChenHoy, 2025-04-08:
This seems like a memory error, what GPU are you running this on and what settings did you use? 

You likely need to shave-off some memory by using less Gaussians, a lower input resolution or trying on a smaller video

zyh778, 2025-04-09:
My GPU is RTX3090, I've fixed the problem by referring to the previous issue and the relevant information I looked up, also, I found that in the setup.py script, the installed diff-gaussian-rasterization comes from the thirdparty/diff-gaussian- rasterization-w-pose folder, so if I want to use MCMC's method, is it sufficient to just modify the relevant folder in setup.py and re-run python setup.py install?

ChenHoy, 2025-04-16:
Hi, I am glad that you could work this out. 

Yeah you are right, that would be sufficient. The setup.py is modified on the mcmc branch though, so if you switch you can also just comment out the modules which you already compiled and only recompile the Gaussian Renderer portion. 

As the issue got resolved, I will close with this comment, thank you!

closed 2025-04-16 by ChenHoy