ChenHoy/DROID-Splat #26: About the malloc(): invalid size (unsorted) during tracking

pengchongH, 2025-05-05:
Thanks for this nice work. I met the issue that malloc(): invalid size (unsorted) during tracking, and if I run it multiple times, this issue won't exist every time. Could you please share some ideas related to this issue. Thanks for your help in advance.

ChenHoy, 2025-05-09:
Hi, so what version of CUDA and PyTorch are you using? 

We sometimes observed similar problems for CUDA 12.1 and think that this is happening when frontend and backend access the video buffer at the same time. 

DROID-SLAM, GO-SLAM and other implementations all use the video.get_lock() method to guard the Processes. I will push another version that uses a true multiprocessing.Lock instead of the multiprocessing.Value used in these default implementations, but I am unsure if this resolves it. We got mostly rid of these by just using a different `sleep_delay` value. 

You could also try to switch to a newer CUDA version, hope that this helps!

ChenHoy, 2025-06-27:
Hi, pushed a version with a true multiprocessing.Lock to guard the CUDA kernel calls during Bundle Adjustment both in Frontend and Backend. If this error persists, we would need to perform changes on the Optimization CUDA code itself or sacrifice parallelism I think.