ChenHoy/DROID-Splat #27: Out of bounds error and TypeError when testing ScanNet scene0000_00

7yzx, 2025-05-23:
Thanks for your great work, But I met some problem when testing **ScenNet** scene0000_00 
I test in cuda11.8, nvcc 11.8 and cuda12.1 ,nvcc 12.4. I just met two probem.

1. First is Out of bounds
I saw your advise in https://github.com/ChenHoy/DROID-Splat/issues/26, but I change sleep_delay to 1 still met this. Can you give me some advice on mapping?
```
Process Frontend Tracking:
Traceback (most recent call last):
  File "/home/yzx/miniconda3/envs/droidsplat_cu12/lib/python3.10/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/yzx/miniconda3/envs/droidsplat_cu12/lib/python3.10/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/farsee2/YZX_code/SLAM/RGB/DROID-Splat_cuda12/src/slam.py", line 327, in tracking
    self.frontend(timestamp, image, depth, intrinsic, gt_pose, static_mask=static_mask)
  File "/home/yzx/miniconda3/envs/droidsplat_cu12/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1518, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/yzx/miniconda3/envs/droidsplat_cu12/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1527, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/yzx/miniconda3/envs/droidsplat_cu12/lib/python3.10/site-packages/torch/utils/_contextlib.py", line 115, in decorate_context
    return func(*args, **kwargs)
  File "/home/yzx/YZX_code/SLAM/RGB/DROID-Splat_cuda12/src/frontend.py", line 42, in forward
    self.optimizer()  # Local Bundle Adjustment
  File "/home/yzx/YZX_code/SLAM/RGB/DROID-Splat_cuda12/src/frontend.py", line 230, in __call__
    self.__update()
  File "/home/yzx/YZX_code/SLAM/RGB/DROID-Splat_cuda12/src/frontend.py", line 157, in __update
    self.video.poses[self.t1] = (dP * SE3(self.video.poses[self.t1 - 1])).vec()
IndexError: index 256 is out of bounds for dimension 0 with size 256
```

2. TypeError: median() received an invalid combination of arguments - got (dim=list, ),
Is this a bug or something? It doesn't seem to affect the program.
```
  6%|███████▊                                                                                                                     | 347/5578 [00:11<01:39, 52.51it/s][Frontend] Running frontend over [11, 19] with 40 factors.
Gaussian Optimization:   6%|██████▎                                                                                                  | 6/100 [00:00<00:10,  9.15it/s]Process Backend:
Traceback (most recent call last):
  File "/home/yzx/miniconda3/envs/droidsplat_cu12/lib/python3.10/multiprocessing/process.py", line 314, in _bootstrap
    self.run()
  File "/home/yzx/miniconda3/envs/droidsplat_cu12/lib/python3.10/multiprocessing/process.py", line 108, in run
    self._target(*self._args, **self._kwargs)
  File "/home/yzx/YZX_code/SLAM/RGB/DROID-Splat_cuda12/src/slam.py", line 463, in global_ba
    self.backend(add_ii=all_loop_ii, add_jj=all_loop_jj)
  File "/home/yzx/miniconda3/envs/droidsplat_cu12/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1518, in _wrapped_call_impl
    return self._call_impl(*args, **kwargs)
  File "/home/yzx/miniconda3/envs/droidsplat_cu12/lib/python3.10/site-packages/torch/nn/modules/module.py", line 1527, in _call_impl
    return forward_call(*args, **kwargs)
  File "/home/yzx/YZX_code/SLAM/RGB/DROID-Splat_cuda12/src/backend.py", line 74, in forward
    _, n_edges = self.optimizer.dense_ba(
  File "/home/yzx/miniconda3/envs/droidsplat_cu12/lib/python3.10/site-packages/torch/utils/_contextlib.py", line 115, in decorate_context
    return func(*args, **kwargs)
  File "/home/yzx/YZX_code/SLAM/RGB/DROID-Splat_cuda12/src/backend.py", line 206, in dense_ba
    self.perform_ba(graph, t_start, t_end, steps, iters, motion_only)
  File "/home/yzx/miniconda3/envs/droidsplat_cu12/lib/python3.10/site-packages/torch/utils/_contextlib.py", line 115, in decorate_context
    return func(*args, **kwargs)
  File "/home/yzx/YZX_code/SLAM/RGB/DROID-Splat_cuda12/src/backend.py", line 146, in perform_ba
    scales_before = self.video.disps[t_start:t_end].median(dim=[1, 2]).clone()
TypeError: median() received an invalid combination of arguments - got (dim=list, ), but expected one of:
 * ()
      didn't match because some of the keywords were incorrect: dim
 * (int dim, bool keepdim)
 * (name dim, bool keepdim)
```

ChenHoy, 2025-05-23:
Hi, I resolved the median error in a later commit, that I can push today :) I suspect the first error on ScanNet is because ScanNet has much longer scenes than others. so a buffer size of 256  for storing the keyframes is too small. I have the following suggestion: 
- I will later push the fix for the median error
- The new paper version will include numbers for ScanNet and some updates for old scenes. For that reason, I have working configs for ScanNet, that you can try directly without changing much
- The latest push will also include KITTI configs, because we have experiments with traditional loop closures on long range scenes. We noticed that for long scenes (ScanNet as well), we can save a lot of memory by simply storing the image buffer in int8 and only convert to float when needed. The new version therefore uses much less memory

7yzx, 2025-05-23:
Thanks a lot ! 
Your work is really great!

ChenHoy, 2025-06-05:
Hey, I am very sorry, but I still need some time and can only make it early next week. I hope I do not make you wait too long :/

WhiteFireFox, 2025-06-10:
> Hi, thank you for your excellent work!
> 
> I'm encountering a `TypeError: median() received an invalid combination of arguments`, which seems to occur during the `global_ba` step. From what I understand, the mapping depends on the completion of `global_ba`, since it checks:
> 
> while (self.tracking_finished + self.backend_finished) < 2 and run:
> If `global_ba` doesn't finish successfully, it seems we can't obtain the final results such as SSIM and LPIPS.
> 
> Could you advise how to fix this issue so that we can reproduce the results reported in the paper, including SSIM and LPIPS?
> 
> Looking forward to your response—thanks again for your great work!

Maybe can we fix it by this? (is it avaliable?)

```
def perform_ba(self, graph: FactorGraph, t_start: int, t_end: int, steps: int, iters: int, motion_only: bool):
        with self.video.get_lock():
            # NOTE chen: computing the scale of a scene is not straight-forward, we simply take the mean disparity as proxy
            scales_before = self.video.disps[t_start:t_end].flatten(start_dim=1).median(dim=1).values.clone()
            poses_before = self.video.poses[t_start + 1 : t_end].clone()  # Memoize pose before optimization

        # Use t_start + 1 to always fix the first pose!
        graph.update_lowmem(t0=t_start + 1, t1=t_end, steps=steps, iters=iters, max_t=t_end, motion_only=motion_only)

        with self.video.get_lock():
            poses_after = self.video.poses[t_start + 1 : t_end]  # Memoize pose before optimization
            scales_after = self.video.disps[t_start:t_end].flatten(start_dim=1).median(dim=1).values.clone()
            # Memoize pose change in self.video so other Processes can adapt their datastructures
            self.accumulate_pose_change(poses_before, poses_after, t0=t_start + 1, t1=t_end)
            # Memoize scale change in self.video so other Processes can adapt their datastructures
            self.accumulate_scale_change(scales_before, scales_after, t0=t_start, t1=t_end)
```

ChenHoy, 2025-06-27:
Ok, all changes are tested and pushed. Feel free to reopen this in case there is something else :)